dron0gus/wideband #14: Dual WB TS CRC check error

Zazins, 2023-08-02:
Hi! 

CRC check error when opening or changing setting, but settings could be set permanent. 
Checked with Bluetooth and Bootloader serial line and same error.  
Firmware and .ini file latest  signature "rusEFI 2023.05.10.wideband_dual"

[image]

D0kRay, 2024-01-30:
I get the same error. No clue why

dron0gus, 2024-03-16:
This is related to `highSpeedOffsets` used in **Full Optimized – High Speed** TunerStudio protocol.
Something is missed in ini file. TunerStudio should not try to Burn this page. `highSpeedOffsets` is `noMsqSave` and also should not be saved in non-volatile memory.

xtitoris, 2026-02-08:
@dron0gus
For some reason noMsqSave does not work as a last attribute, maybe a bug in TunerStudio.
However, if added in `[ConstantsExtension]` section, it works correctly.

```
[ConstantsExtension]
    noMsqSave = highSpeedOffsets
```

dron0gus, 2026-10-29:
Fixed in https://github.com/dron0gus/wideband/commit/617a3973ca92efcaa5a9e44ae3733f1992a76b3a

closed 2026-10-29 by dron0gus