dron0gus/wideband #19: NTK 55214916

rusefillc, 2024-07-18:
What on earth is https://www.ngk.com/ntk-24394-oe-identical-afr-sensor?

5 wire

Fiat 500 Abarth 1.4T Turbo Front O2 Pre Cat Oxygen Sensor Genuine Oem 2012-2018

[image]

[image]

rusefillc, 2024-07-18:
Reminds me of https://github.com/rusefi/hellen154hyundai/issues/143

rusefillc, 2024-07-18:
Here's another monstrosity

250-25062
2010-2022 kia Hyundai

[image]

ElDominio, 2025-08-13:
https://www.bmotorsports.com/shop/product_info.php/cPath/103_107/products_id/3295 ntk controller @rusefillc

rusefillc, 2025-08-13:
@ElDominio do we know if mustang uses ZFAS-D, ZFAS-U2 or ZFASU3? do we know which one bmotorsports controls? do they have same exact plug?